pothosware/SoapyRTLSDR #4: compile test on msvc

guruofquality, 2015-09-23:


guruofquality, 2015-09-25:
@cjcliffe One issue came up for me when building on MSVC. It didn't have a template overload for using isnan on an integer. Many of the args parsing calls use stoi, which through throw invalid_argument if the input is bad. Here are the changes, it hasnt broken anything for me, just running it by you:

* https://github.com/pothosware/SoapyRTLSDR/commit/391a418fb7c73a807f35e051c9e98c5d19025eb4
* https://github.com/pothosware/SoapyRTLSDR/commit/bbfed3a7081894a4ca44b8595da95e813f0d124f

cjcliffe, 2015-09-25:
:+1: No complaints here; I haven't really used std::stoi and std::isnan before -- I just happened upon those while looking up simpler C++ replacements to some C stuff I was using.   Also makes sense to use the single char for checking IF number since it's fixed at length 6.

Haven't had a chance to get SoapySDR running on my windows box yet; I will likely do that this weekend as I want to get some CubicSDR Soapy supported test builds out.

guruofquality, 2015-09-26:
* and after that, it now builds on MSVC 2012 and 2013

closed 2015-09-26 by guruofquality