pothosware/SoapyRTLSDR #8: Issue with sync API

guruofquality, 2015-09-28:
Referencing this issue: https://github.com/pothosware/SoapyRemote/issues/9 I noticed that a FM demod demo was extremely choppy and causing audio underflows. This didnt happen with the RTL support from SoapyOsmo.

So I switched to the async api on this branch and the problem went away: https://github.com/pothosware/SoapyRTLSDR/commits/async_stream

I suspect a bug in the sync API, and I wonder if anyone else is seeing a similar issue. Although the sync API is more fitting here, if its going to have issues, the async_stream branch needs some cleanup and can be merged.

cjcliffe, 2015-09-28:
I would say we should go ahead and pull the async version to master; there's no downside in performance that I've found; the problems seem to have just evaporated.

guruofquality, 2015-09-29:
The async work is now merged into master and I did some cleanup as well.

* The timeouts are implemented
* The overflows are indicated
* The thread is started/stopped using the activate/deactivate hooks

cjcliffe, 2015-09-29:
Excellent; thanks -- with this update it's much more stable on SoapyRemote than rtl_tcp.  
rtl_tcp constantly crashed on my Pi with the slightest interruption, could barely handle 2Mhz and I was never able to re-load until I restarted the Pi..

closed 2015-09-29 by cjcliffe