pothosware/SoapyRTLSDR #11: Direct buffer access calls for RX

guruofquality, 2015-10-08:
Ditto for the HackRF, more details here: https://github.com/pothosware/SoapyHackRF/issues/9

* acquire and release buffer using the queue's head and tail pointers
* implement readStream using  acquire and release calls

cjcliffe, 2015-10-17:
Updates look good, thanks!   I'll see what I can do to implement the same in SoapySDRPlay.

guruofquality, 2015-10-18:
For the hackrf, I will probably end up making used of the Stream* pointer to house some stream related data structures. In this case it avoids some code duplication rather than specifying fifo counters, buffers, threads, etc twice for both RX and TX.

Not sure about doing this for RTL as well since it only is a single channel device. Maybe just for correctness or to stand as an example. Anyway, just bringing it up since you are considering re-factoring SoapySDRPlay.

```
struct SoapyRTLStream
{
  //threads
  //buffers
  //fifo counts
  //async handler
  //overflow indicator
};

Stream *setupStream(...)
{
    auto data = new SoapyRTLStream();
    //stream setup
    return (Stream *)data;
}

void closeStream(Stream * stream)
{
    auto data = (SoapyRTLStream*)stream;
    delete data;
}

```

guruofquality, 2016-04-25:
Also done a while ago. Closing old issue.

closed 2016-04-25 by guruofquality