pothosware/SoapyRTLSDR #20: setBandwidth/getBandwidth no-ops, should call rtlsdr_set_and_get_tuner_bandwidth?

rxseger, 2016-07-17:
SoapyRTLSDR's bandwidth methods currently only call the stubs in the superclass, which return 0 Hz:

https://github.com/pothosware/SoapyRTLSDR/blob/a44abc3f1ac2c62d707902551f1c2c1e70551c84/Settings.cpp#L448-L463
https://github.com/pothosware/SoapySDR/blob/master/lib/Device.cpp#L486-L499

should these methods be bridged to librtlsdr's rtlsdr_set_and_get_tuner_bandwidth() (and/or rtlsdr_set_tuner_bandwidth()??

Note that these librtlsdr APIs to get/set bandwidth are relatively new, not all installations may necessarily have it yet, so it may be worth holding off until it can be confirmed it is widely available as needed. I have no immediate need for it, just noticed this gap in librtlsdr->SoapyRTLSDR bridging while working on something else.

References:

* New functions rtlsdr_set_tuner_bandwidth()  https://github.com/librtlsdr/librtlsdr/commit/92df068dc5acae9e671eefe1f80626319819c52b
* [PATCH] add get/set BW for rtl-sdr http://lists.osmocom.org/pipermail/osmocom-sdr/2015-June/000077.html
* Bug report in another project where there were issues depending on bandwidth setting functions in librtlsdr: https://github.com/jpoirier/gortlsdr/issues/14

guruofquality, 2021-01-25:
This issue is old enough, im sure rtlsdr_set_tuner_bandwidth is mainline code now :-)

Anyone want to attempt a PR to close it off?

zuckschwerdt, 2021-01-25:
On it. If there are no blockers I'll commit in an hour or so.

zuckschwerdt, 2021-01-25:
Implemented, tested working. Someone still needs to add more sensible values for listBandwidths(), getBandwidthRange() please ;)

commit b5489df7 (2021-01-25): Add setBandwidth/getBandwidth implementation (closes #20)
closed 2021-01-25 by zuckschwerdt