pothosware/SoapyRTLSDR #21: Automatic gain mode setting, rtlsdr_set_tuner_gain_mode(dev, 0) equivalent

rxseger, 2016-07-17:
librtlsdr implements `rtlsdr_set_tuner_gain_mode()` [here](https://github.com/librtlsdr/librtlsdr/blob/aabe859851c8cd5471adcf68cea527a08acb369a/src/librtlsdr.c#L1063) for setting "automatic" (0) or "manual" (1) gain mode, this API doesn't appear to be bridged in SoapyRTLSDR.

Should it be, or how can one enable auto gain? (curious how CubicSDR does it). Without setting some gain, I can only receive silence, using 40 dB as a default for now but having the hardware/driver automatically choose the default gain values seems to be preferable, especially if it can differ between devices or other configurations.

(Note, mainly to self: automatic gain is distinct from automatic gain *correction* (agc), librtlsdr's `rtlsdr_set_agc_mode()`, which *is* bridged in SoapyRTLSDR setGainMode(bool). But not https://github.com/pothosware/SoapyRTLSDR/search?q=rtlsdr_set_tuner_gain_mode)

cjcliffe, 2016-07-17:
Not sure; I'll take a look at the gain functions again -- at the time of writing I don't recall there being a second option -- AGC seems to be functional here so I'm not sure if this is just a more recent development?    

CubicSDR has only used/bundled git://git.osmocom.org/rtl-sdr.git to date as it is still considered the official source to my knowledge.

rxseger, 2016-07-17:
Ah, git://git.osmocom.org/rtl-sdr.git is the official librtlsdr not https://github.com/librtlsdr/librtlsdr? Too many forks, confusing :/

But rtlsdr_set_tuner_gain_mode is in osmocom/rtl-sdr as well, was added back in 2012: https://github.com/librtlsdr/librtlsdr/commit/86c34428aabebc79ac0de7f1194e6b755a6fa213. The librtlsdr rtl_* tools will enable auto gain mode if no `-g` command-line argument is specified (`verbose_auto_gain()` -> `rtlsdr_set_tuner_gain_mode(dev, 0);`). With `-g #`, they'll enable manual gain mode and set the tuner gain (`verbose_gain_set()` -> `rtlsdr_set_tuner_gain_mode(dev, 1); rtlsdr_set_tuner_gain(dev, gain);`). AGC can be enabled/disabled independently of auto/manual gain mode, I believe.

cjcliffe, 2016-07-17:
@rxseger ah, I see it now -- I'll try that here soon unless you work up a PR in the meantime I can merge; thanks for catching that 👍

zuckschwerdt, 2018-10-28:
I'd like to add [rtlsdr_set_tuner_gain_mode()](https://github.com/osmocom/rtl-sdr/blob/master/include/rtl-sdr.h#L253) for automatic tuner gain.
What would be the best way to fit that into the SoapySDR API? Is a magic value for [setGain()](https://github.com/pothosware/SoapyRTLSDR/blob/master/Settings.cpp#L265) acceptable (i.e. extend the gain range to 41dB and use that as switch)?
Or should I add the switch to the writeSetting/readSetting facility?
I guess adding a new gain element TUNER_AUTO=0/1 could be done too but seems obscure.

zuckschwerdt, 2018-10-28:
I can't really say if people like to use AGC, for me it's always off and the tuner gain mode to auto.
I'll code the setting option and run some tests.

zuckschwerdt, 2018-10-29:
Maybe I'm wrong but the "AGC" with `rtlsdr_set_agc_mode()` is a digital gain stage and not too useful. The actual switch between manual gain and automatic gain in the tuner (analog) side is `rtlsdr_set_tuner_gain_mode()`.
I know it's a hard decision to break existing interfaces and will cause grief.
Nevertheless I need to propose, `setGainMode()` should switch automatic gain on the tuner as `setGain()` sets a manual gain there. And the digital AGC should be an option switch hidden away with the settings :)
That said I'm fine with the (current) other way around, as long as I can access the features.

zuckschwerdt, 2018-10-29:
I have outlined both ideas in #32 and #33.

guruofquality, 2018-11-03:
Cool, the PR is merged, I will update the changelog

closed 2018-11-03 by guruofquality