pothosware/SoapyRTLSDR #23: Adding function to set GPIO pins to the SoapyRTL Driver?

bcattle, 2016-09-18:
I want to add a GPIO interface to the Soapy RTL-SDR driver. I'm designing an automatic antenna tuner that I want to control using the GPIO pins on the RTL-SDR. 

The Blade RF Soapy driver has a [GPIO interface](https://github.com/pothosware/SoapyBladeRF/blob/master/bladeRF_SoapySDR.hpp#L217). I'd like to implement a similar interface for the RTL-SDR:

    void writeGPIO(const std::string &bank, const unsigned value);
    void writeGPIO(const std::string &bank, const unsigned value, const unsigned mask);

The issue is that the librtlsdr driver supports these changing these pins with [`rtlsdr_set_gpio_bit`](https://github.com/steve-m/librtlsdr/blob/8b4d755ba1b889510fba30f627ee08736203070d/src/librtlsdr.c#L556), but this function is not public by default. 

So I have a couple questions:

1. On the Soapy side, does this API make sense?
2. What should I do about making that function public in the librtlsdr driver? Is there a way to check whether that function is available, either in a future version of the driver or a custom fork? 

Thanks.

cjcliffe, 2016-09-18:
@bcattle interesting; I wasn't aware the RTL-SDR even had GPIO let alone 8 of them.

I'll leave this here for reference: http://lea.hamradio.si/~s57uuu/mischam/rtlsdr/ports.html as it appears to show what's needed; adding the prototypes to link in the functions or adding them ourselves should be trivial.

I'd be interested in modifying one of the dongles I have here to test and implement this.

bcattle, 2016-09-19:
I have the v.3 dongle, and it has 4 of the pins brought out to pads, see http://imgur.com/a/XWGQt. The pads are labelled 29, 30, 31, and 32 corresponding to GPIO P5, P4, P2 and P1 according to http://lea.hamradio.si/~s57uuu/mischam/rtlsdr/ports.html.  

I merged the Marko Cebokli / S57UUU code into a forked version of the driver here: https://github.com/bcattle/rtl-sdr. It compiles, but I haven't tested it yet.

guruofquality, 2016-09-19:
> What should I do about making that function public in the librtlsdr driver? Is there a way to check whether that function is available when compiling, either in a future version of the driver or a custom fork? 

Take a look at this [CMakeLists](https://github.com/pothosware/SoapyUHD/blob/master/CMakeLists.txt#L62) which searches the header file for features and sets a define for the C++ source to use with ifdef. So basically the SoapyRTL could support the litany of feature branches out there, but always conditional check based on what its compiled against.

bcattle, 2016-09-19:
If anyone wants to play with this, I added a command line utility to read and write the GPIO pins to a fork of the driver at https://github.com/bcattle/rtl-sdr. I tested it with my dongle and was able to set pins. 

To use, run it with `-s pin=val`. Running it without args just prints the value of the register.

```
$ ./rtl_gpio
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
GPIO byte is set to: 0x8 (0b1000)

$ ./rtl_gpio -s 1=1
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Setting pin 1 to 0b1
GPIO byte is set to: 0xa (0b1010)
```

On the Soapy side I [added functions](https://github.com/bcattle/SoapyRTLSDR/commit/75a07479db36216c6e1b5e934f584d251dd484d1) to set the pins to https://github.com/bcattle/SoapyRTLSDR.

guruofquality, 2016-09-19:
> On the Soapy side I added functions to set the pins to https://github.com/bcattle/SoapyRTLSDR. 

FYI, the GPIO functions need to [exactly match the ones in SoapySDR::Device](https://github.com/pothosware/SoapySDR/blob/master/include/SoapySDR/Device.hpp#L1111) for the overloading to work.

bcattle, 2016-09-21:
Got it. I'll update. Thanks