pothosware/SoapyRTLSDR #24: Feature request: implement r820tweak

dc1rdb, 2016-12-12:
Would it be possible to implement the features of https://github.com/gat3way/r820tweak into this driver? r820tweak is a modified RTL-SDR driver that exposes the R820T2 device gain stages and filters and makes them accessible through a GUI app.

darksidelemm, 2020-03-07:
+1 on this. Being able to adjust more than just one gain setting is important in obtaining the best performance out of these devices. The Airspy drivers use optimised combinations of the VGA, Mixer, and LNA gain settings for their 'Linearity' and 'Sensitivity' gain profiles, which are defined here: https://github.com/airspy/airspyone_host/blob/master/libairspy/src/airspy.c#L117

The R820Tweak code which sets the relevant registers is here: https://github.com/gat3way/r820tweak/blob/master/src/tuner_r82xx.c#L1076

zuckschwerdt, 2020-03-07:
Sounds interesting. Note that SoapyRTLSDR uses librtlsdr as API (`rtlsdr_` calls) the e.g. `r82xx_write_reg_mask` calls aren't available. The tweaks should be merged to the official librtlsdr first, then we could use them as API.

eabase, 2025-02-20:
Interesting. Was this ever merged?

zuckschwerdt, 2025-02-20:
You would need to lookup if any of the librtlsdr projects supports this as API.
https://github.com/osmocom/rtl-sdr/
https://github.com/librtlsdr/librtlsdr
or maybe other forks.