pothosware/SoapyRTLSDR #30: Feature Request: Add toggle for v3 dongle sw controlled bias-t

unixpunk, 2017-12-13:
Hi, would be a cool feature to be able to use v3 sw controlled bias-t.  https://github.com/rtlsdrblog/rtl_biast/blob/master/src/rtl_biast.c  Happy to test!  Thanks!

######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Probe device driver=rtlsdr
Found Rafael Micro R820T tuner
Found Rafael Micro R820T tuner

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=RTLSDR
  hardware=RTLSDR
  origin=https://github.com/pothosware/SoapyRTLSDR
  rtl=0

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * Direct Sampling - RTL-SDR Direct Sampling Mode
       [key=direct_samp, default=0, type=string, options=(0, 1, 2)]
     * Offset Tune - RTL-SDR Offset Tuning Mode
       [key=offset_tune, default=false, type=bool]
     * I/Q Swap - RTL-SDR I/Q Swap Mode
       [key=iq_swap, default=false, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS8, CS16, CF32
  Native format: CS8 [full-scale=128]
  Stream args:
     * Buffer Size - Number of bytes per buffer, multiples of 512 only.
       [key=bufflen, units=bytes, default=262144, type=int]
     * Ring buffers - Number of buffers in the ring.
       [key=buffers, units=buffers, default=15, type=int]
     * Async buffers - Number of async usb buffers (advanced).
       [key=asyncBuffs, units=buffers, default=0, type=int]
  Antennas: RX
  Full gain range: [0, 49.6] dB
    TUNER gain range: [0, 49.6] dB
  Full freq range: [23.999, 1764] MHz
    RF freq range: [24, 1764] MHz
    CORR freq range: [-0.001, 0.001] MHz
  Sample rates: 0.25, 1.024, 1.536, 1.792, 1.92, 2.048, 2.16, 2.56, 2.88, 3.2 MSps

jketterl, 2020-03-14:
I'm probably not the best guy to do this, but I still tried: https://github.com/jketterl/SoapyRTLSDR/tree/biastee_support

I don't even dare to open a pull request since I'm unable to test it. Is there somebody that could confirm that it's working?

zuckschwerdt, 2020-03-14:
@jketterl your code looks good. Do open a PR :)

[SoapyAirspy](https://github.com/pothosware/SoapyAirspy/blob/master/Settings.cpp#L369) calls this `"biastee"`, and [SoapySDRPlay](https://github.com/pothosware/SoapySDRPlay/blob/master/Settings.cpp#L915) calls it `"biasT_ctrl"` perhaps we could use one of those name to help apps discover this setting. (Though `"bias_tee"` would have been a better name, in my opinion…)

Would be great if somebody with this mod could test.

jketterl, 2020-03-14:
You're right, it's not really helpful to introduce a third key. I'll change it to the airspy way, simply because it's easier to read.

jketterl, 2020-05-09:
just went through some old notifications... since the PR is open, this should be resolved now, too, shouldn't it?

closed 2020-05-09 by guruofquality