pothosware/SoapyRTLSDR #38: Device init sometimes does not work (RTL SDR V3)

dmitryelj, 2019-05-25:
Hi all,

Thanks for the nice product.
In the last version there is an issue with access to RTLSDR - sometimes it initializing correctly and sometimes is not.

[image]

Sometimes it works, and sometimes is not, the probability is about 50% :) Symptoms: the device serial number is empty, the gains are not correct (both are zeros).
It also affects CubicSDR - it does not work correctly.

Device RTL SDR V3, system Windows 10, the drivers were installed with PothosSDR-2019.03.24-vc14-x64.exe, in old version PothosSDR-2018.09.09-vc14-x64.exe all ok.

Code to reproduce: 
```
import SoapySDR

print("SDR devices:")
for d in SoapySDR.Device.enumerate(''):
        print(d)
print()

print("SDR device gains:")
device = SoapySDR.Device({"driver": "rtlsdr"})
for gain in device.listGains(SoapySDR.SOAPY_SDR_RX, 0):
        print("  %s: %s" % (gain, device.getGainRange(SoapySDR.SOAPY_SDR_RX, 0, gain)))
print()
```

vsonnier, 2019-05-25:
> Addition: if I replace SoapySDR.Device.enumerate('') to SoapySDR.Device.enumerate('driver=rtlsdr') it works always correct, I don't know the reason.

Indeed CubicSDR use the generic version `SoapySDR::Device::enumerate()` most of the time. Now you can emulate your workaround in CubicSDR by setting a device manually:

In the SDR Devices window --> Add , then select 'rtlsdr', do not add anu other arguments (driver=rtlsdr is automatically passed) and press OK.
The device then appears in the `Manual` tree, selects this one and run.

guruofquality, 2019-05-26:
> Addition: if I replace SoapySDR.Device.enumerate('') to SoapySDR.Device.enumerate('driver=rtlsdr') it works always correct, I don't know the reason.

I think the main difference is that enumerate without driver=rtlsdr tries to call the discovery function of the other supported modules. I wonder if some other support module could be causing an issue...

One test I have in mind, run `SoapySDRUtil --info`, this will give a list of support modules. Delete one at a time until the problem goes away. Perhaps the last module to be deleted will be the one causing a problem.

If it comes down to one module left, but it doesnt matter which one, it might be a threading issue instead, since discovery is done in parallel, one thread per support module.

Just a guess...

dmitryelj, 2019-05-26:
Thanks a lot for the hint, it works.

I had this list of drivers:
```
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/airspyhfSupport.dll (0.1.0-e8582d4)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/airspySupport.dll   (0.1.2-99756be)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/audioSupport.dll    (0.1.0-cf50c7c)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/bladeRFSupport.dll  (0.4.1-1c1e8aa)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/HackRFSupport.dll   (0.3.3-3c514ce)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/IrisSupport.dll     (2019.01.0.2-7cdbc81)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/LMS7Support.dll     (19.01.0-8bf4230b)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/miriSupport.dll     (0.2.6-ffa0a7e)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/osmosdrSupport.dll  (0.2.6-ffa0a7e)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/PlutoSDRSupport.dll (0.1.1-b0203fd)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/RedPitaya.dll       (0.1.1-3d576f8)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/remoteSupport.dll   (0.5.2-a6fbf2d)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/rtlsdrSupport.dll   (0.3.1-c8c5b92)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/sdrPlaySupport.dll  (0.3.0-31a3a36)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/uhdSupport.dll      (0.3.5-37ded63)
```

Then I removed ones, I'm not using, and the problem is gone.
My current drivers list looks like this:
```
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/airspyhfSupport.dll (0.1.0-e8582d4)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/audioSupport.dll    (0.1.0-cf50c7c)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/LMS7Support.dll     (19.01.0-8bf4230b)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/miriSupport.dll     (0.2.6-ffa0a7e)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/osmosdrSupport.dll  (0.2.6-ffa0a7e)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/remoteSupport.dll   (0.5.2-a6fbf2d)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/rtlsdrSupport.dll   (0.3.1-c8c5b92)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/sdrPlaySupport.dll  (0.3.0-31a3a36)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/uhdSupport.dll      (0.3.5-37ded63)
```

dmitryelj, 2019-05-26:
Addition: I sorted files by creation day, and started removing the newest ones.

Verdict: the guilty is "PlutoSDRSupport.dll". This ticket can be closed.

closed 2019-05-26 by dmitryelj