pothosware/SoapyRTLSDR #40: undefined reference to `__atomic_load_8'

n8twj, 2019-08-06:
I cannot seem to compile SoapyRTLSDR on my raspberry PI.   I have an RTL SDR v3.  I have followed steps in the various Wiki pages to install SoapySDR and SoapyRemote. 

raspberrypi:~/SoapyRTLSDR/build# cat /etc/debian_version
10.0

# cat /proc/cpuinfo 
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a02082
Serial		: 0000000058407487

raspberrypi:~/SoapyRTLSDR/build# rtl_test
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000010

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner

raspberrypi:~/SoapyRTLSDR/build# cmake ..
-- The CXX compiler identification is GNU 8.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type not specified: defaulting to release.
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Checking for module 'librtlsdr'
--   Found librtlsdr, version 0.6.0-5-g8183
-- Found rtlsdr: /usr/local/lib/librtlsdr.so  
-- RTLSDR_INCLUDE_DIRS - /usr/local/include
-- RTLSDR_LIBRARIES - /usr/local/lib/librtlsdr.so
-- Performing Test HAS_STD_CXX11
-- Performing Test HAS_STD_CXX11 - Success
-- Found Git: /usr/bin/git (found version "2.20.1") 
-- Module rtlsdrSupport configured with version: 0.3.1-5c5d950
-- Configuring done
-- Generating done
-- Build files have been written to: /root/SoapyRTLSDR/build
raspberrypi:~/SoapyRTLSDR/build# make
Scanning dependencies of target rtlsdrSupport
[ 20%] Building CXX object CMakeFiles/rtlsdrSupport.dir/Version.cpp.o
[ 40%] Linking CXX shared module librtlsdrSupport.so
/usr/bin/ld: CMakeFiles/rtlsdrSupport.dir/Settings.cpp.o: in function `SoapyRTLSDR::getHardwareTime(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const':
Settings.cpp:(.text+0x12c): undefined reference to `__atomic_load_8'
/usr/bin/ld: CMakeFiles/rtlsdrSupport.dir/Settings.cpp.o: in function `SoapyRTLSDR::setHardwareTime(long long, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
Settings.cpp:(.text+0x174): undefined reference to `__atomic_store_8'
/usr/bin/ld: CMakeFiles/rtlsdrSupport.dir/Settings.cpp.o: in function `SoapyRTLSDR::setSampleRate(int, unsigned int, double)':
Settings.cpp:(.text+0x5b4): undefined reference to `__atomic_load_8'
/usr/bin/ld: Settings.cpp:(.text+0x62c): undefined reference to `__atomic_store_8'
/usr/bin/ld: CMakeFiles/rtlsdrSupport.dir/Settings.cpp.o: in function `SoapyRTLSDR::SoapyRTLSDR(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)':
Settings.cpp:(.text+0x4ca4): undefined reference to `__atomic_store_8'
/usr/bin/ld: CMakeFiles/rtlsdrSupport.dir/Streaming.cpp.o: in function `SoapyRTLSDR::rx_callback(unsigned char*, unsigned int)':
Streaming.cpp:(.text+0x29f0): undefined reference to `__atomic_fetch_add_8'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/rtlsdrSupport.dir/build.make:131: librtlsdrSupport.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/rtlsdrSupport.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

guruofquality, 2019-08-06:
Can you pass `-DOTHER_LIBS="-latomic"` to cmake? You might try "-pthread" as well, or try a few other libs. I assume its just a simple linker error and some usual dependency isnt being pull in by default. Let us know what you find. THanks!

n8twj, 2019-08-06:
Hi - Looks liked that worked.  I now have SoapyRemote up and running in CubicSDR with the dongle connected to the rPI 

raspberrypi:~/SoapyRTLSDR/build# cmake ../ -DOTHER_LIBS="-latomic"
-- Build type not specified: defaulting to release.
-- RTLSDR_INCLUDE_DIRS - /usr/local/include
-- RTLSDR_LIBRARIES - /usr/local/lib/librtlsdr.so
-- Module rtlsdrSupport configured with version: 0.3.1-5c5d950
-- Configuring done
-- Generating done
-- Build files have been written to: /root/SoapyRTLSDR/build
raspberrypi:~/SoapyRTLSDR/build# make
Scanning dependencies of target rtlsdrSupport
[ 20%] Building CXX object CMakeFiles/rtlsdrSupport.dir/Version.cpp.o
[ 40%] Linking CXX shared module librtlsdrSupport.so
[100%] Built target rtlsdrSupport

takurx, 2019-10-11:
Hi I also had been caught in a same trap on RPi. :-)
I add in Cmakelist.tx. And make, it's ok.
```
set(CMAKE_CXX_LINK_FLAGS "${CMAKE_CXX_LINK_FLAGS} -latomic")
```
I referenced following.
Ref. https://github.com/alexa/avs-device-sdk/issues/1404

guruofquality, 2019-11-16:
You can pass `-DOTHER_LIBS=atomic` to cmake. I would also be happy to accept a PR for adding atomic when the right arm platform is detected.

Perhaps on arm32 some of the atomics are implemented with locks and other funny business so they put the private implementation a library that needs to be linked in -- usually its inline functions.

closed 2019-11-16 by guruofquality

wa2zkd, 2020-01-10:
I had same issue with Buster and now have it working, see open issue

zuckschwerdt, 2020-01-11:
Duplicate issue is #41 with some code in #42. The branch needs some refinement but should already work (only tested for Buster).

LoLLoSeNSo, 2021-12-06:
> Hi I also had been caught in a same trap on RPi. :-) I add in Cmakelist.txt. And make, it's ok.
> 
> ```
> set(CMAKE_CXX_LINK_FLAGS "${CMAKE_CXX_LINK_FLAGS} -latomic")
> ```
> 
> I referenced following. Ref. [alexa/avs-device-sdk#1404](https://github.com/alexa/avs-device-sdk/issues/1404)

I'm using a Raspberry pi 3 with Debian Buster
I tried adding this line in  avs-device-sdk/CMakeLists.txt but I'm having the same issue:

[ 14%] Linking CXX executable InMemoryCommunicationInvokeHandlerTest
/usr/bin/ld: ../../../AVSCommon/libAVSCommon.so: undefined reference to `__atomic_fetch_add_8'
/usr/bin/ld: ../../../AVSCommon/libAVSCommon.so: undefined reference to `__atomic_store_8'
/usr/bin/ld: ../../../AVSCommon/libAVSCommon.so: undefined reference to `__atomic_load_8'

zuckschwerdt, 2021-12-06:
Feel free to suggest using my (cmake) code from #42, it's under "LLVM Release License".