pothosware/SoapyRTLSDR #41: RPi Buster succcess??

wa2zkd, 2020-01-04:
I'm having trouble getting it to build on a Rpi with the latest Buster release. cmake runs OK but at make time I get... atomic 8 issue?

[100%] Linking CXX shared module librtlsdrSupport.so
/usr/bin/ld: CMakeFiles/rtlsdrSupport.dir/Settings.cpp.o: in function `SoapyRTLSDR::getHardwareTime(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const':
Settings.cpp:(.text+0x12c): undefined reference to `__atomic_load_8'
/usr/bin/ld: CMakeFiles/rtlsdrSupport.dir/Settings.cpp.o: in function `SoapyRTLSDR::setHardwareTime(long long, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
Settings.cpp:(.text+0x174): undefined reference to `__atomic_store_8'
/usr/bin/ld: CMakeFiles/rtlsdrSupport.dir/Settings.cpp.o: in function `SoapyRTLSDR::setSampleRate(int, unsigned int, double)':
Settings.cpp:(.text+0x5b4): undefined reference to `__atomic_load_8'
/usr/bin/ld: Settings.cpp:(.text+0x62c): undefined reference to `__atomic_store_8'
/usr/bin/ld: CMakeFiles/rtlsdrSupport.dir/Settings.cpp.o: in function `SoapyRTLSDR::SoapyRTLSDR(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)':
Settings.cpp:(.text+0x4ca4): undefined reference to `__atomic_store_8'
/usr/bin/ld: CMakeFiles/rtlsdrSupport.dir/Streaming.cpp.o: in function `SoapyRTLSDR::rx_callback(unsigned char*, unsigned int)':
Streaming.cpp:(.text+0x29f0): undefined reference to `__atomic_fetch_add_8'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/rtlsdrSupport.dir/build.make:131: librtlsdrSupport.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/rtlsdrSupport.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

guruofquality, 2020-01-04:
Basically the atomic stuff in C++ is usually header only unless it has to be emulated on certain platforms in which case the calls go in a library with locks around them or something like that. Probably because of the 64-bit atomic used here.

You should be able to pass -DOTHER_LIBS=atomic to cmake manually to fix this. But if you are able to test it, I would appreciate a PR that would automatically detect this platform and add atomic to the libraries. :-)

zuckschwerdt, 2020-01-05:
I can prepare a branch with a cmake test for this if you are willing to poke around and test it a few times @wa2zkd

wa2zkd, 2020-01-05:
I'm willing to try, I may not get to it for a few days

zuckschwerdt, 2020-01-05:
Ok, something to get started is in #42. Maybe add
```
message(STATUS "HAVE_CXX_ATOMICS_WITHOUT_LIB - ${HAVE_CXX_ATOMICS_WITHOUT_LIB}")
message(STATUS "HAVE_CXX_ATOMICS64_WITHOUT_LIB - ${HAVE_CXX_ATOMICS64_WITHOUT_LIB}")
```
below `include(CheckAtomic)` in the CMakeLists.txt to see what gets detected.

wa2zkd, 2020-01-06:
I built the new files in this branch, did not complete.   Keep in mind, I am not a programmer!!

 [100%] Linking CXX shared module librtlsdrSupport.so
/usr/bin/ld: CMakeFiles/rtlsdrSupport.dir/Settings.cpp.o: in function `SoapyRTLSDR::getHardwareTime(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const':
Settings.cpp:(.text+0x12c): undefined reference to `__atomic_load_8'
/usr/bin/ld: CMakeFiles/rtlsdrSupport.dir/Settings.cpp.o: in function `SoapyRTLSDR::setHardwareTime(long long, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
Settings.cpp:(.text+0x174): undefined reference to `__atomic_store_8'
/usr/bin/ld: CMakeFiles/rtlsdrSupport.dir/Settings.cpp.o: in function `SoapyRTLSDR::setSampleRate(int, unsigned int, double)':
Settings.cpp:(.text+0x5b4): undefined reference to `__atomic_load_8'
/usr/bin/ld: Settings.cpp:(.text+0x62c): undefined reference to `__atomic_store_8'
/usr/bin/ld: CMakeFiles/rtlsdrSupport.dir/Settings.cpp.o: in function `SoapyRTLSDR::SoapyRTLSDR(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)':
Settings.cpp:(.text+0x4ca4): undefined reference to `__atomic_store_8'
/usr/bin/ld: CMakeFiles/rtlsdrSupport.dir/Streaming.cpp.o: in function `SoapyRTLSDR::rx_callback(unsigned char*, unsigned int)':
Streaming.cpp:(.text+0x29f0): undefined reference to `__atomic_fetch_add_8'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/rtlsdrSupport.dir/build.make:131: librtlsdrSupport.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/rtlsdrSupport.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

zuckschwerdt, 2020-01-06:
I updated the branch. You should now see something like:
```
-- HAVE_CXX_ATOMICS_WITHOUT_LIB -
-- HAVE_CXX_ATOMICS64_WITHOUT_LIB - 1
```
and the build is forced to use the atomic lib. Can you confirm that work and what the two lines above are?

wa2zkd, 2020-01-06:
pi@raspberrypi:~/Dev/SoapyRTLSDR/build $ cmake ..
-- The CXX compiler identification is GNU 8.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type not specified: defaulting to release.
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Checking for module 'librtlsdr'
--   Found librtlsdr, version 0.6.0-11-gb5af
-- Found rtlsdr: /usr/local/lib/librtlsdr.so  
-- RTLSDR_INCLUDE_DIRS - /usr/local/include
-- RTLSDR_LIBRARIES - /usr/local/lib/librtlsdr.so
-- Performing Test HAS_STD_CXX11
-- Performing Test HAS_STD_CXX11 - Success
-- Found Git: /usr/bin/git (found version "2.20.1") 
-- Module rtlsdrSupport configured with version: 0.3.1-5c5d950
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Dev/SoapyRTLSDR/build

I am not seeing any ref to ATOMIC anywhere except in the errors I see when I run make.

wa2zkd, 2020-01-06:
Perhaps an error by me. I thought I could go to the branch and use git clone but I see that grabs the master, not the branch

wa2zkd, 2020-01-06:
So now I got the test working, I think and it says compiler needs libatomic but can't find it. I'll have to look more what is installed here but pretty sure it is

wa2zkd, 2020-01-06:
this is installed:
libatomic1 is already the newest version (8.3.0-6+rpi1).

zuckschwerdt, 2020-01-07:
Ok, so https://packages.debian.org/buster/libatomic1 should have the files in the compiler path (e.g. https://packages.debian.org/buster/armhf/libatomic1/filelist ) -- you are running armhf or is it arm64 (use `uname -a`)?
What did those two lines say?

wa2zkd, 2020-01-07:
Linux raspberrypi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux

It is the debian buster image from the raspian site

wa2zkd, 2020-01-07:
find found 

/usr/lib/arm-linux-gnueabihf/libatomic_ops.a
/usr/lib/arm-linux-gnueabihf/libatomic.so.1
/usr/lib/arm-linux-gnueabihf/libatomic_ops_gpl.a
/usr/lib/arm-linux-gnueabihf/libatomic.so.1.2.0
/usr/lib/arm-linux-gnueabihf/libatomic_ops.la
/usr/lib/arm-linux-gnueabihf/libatomic_ops_gpl.la
/usr/lib/gcc/arm-linux-gnueabihf/8/libatomic.so
/usr/lib/gcc/arm-linux-gnueabihf/8/libatomic.a

zuckschwerdt, 2020-01-07:
Looks like I need to set up a Raspberry Pi with Buster to debug this further. Will take some weeks. Any help welcome.

wa2zkd, 2020-01-07:
Thanks so much for all your help so far...   I'll continue to try things here

wa2zkd, 2020-01-07:
I just completed a fresh install, this time using the degian buster "full image" rather than lite like before. Got the same results. Also ran this 

@raspberrypi:/# ldconfig -p | grep libatomic
	libatomic.so.1 (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libatomic.so.1

wa2zkd, 2020-01-07:
I also found this with reference to atomic issues and buster
https://www.pyimagesearch.com/2019/09/16/install-opencv-4-on-raspberry-pi-4-and-raspbian-buster/

wa2zkd, 2020-01-08:
from CmakeError.log
Performing C++ SOURCE FILE Test HAVE_CXX_ATOMICS64_WITHOUT_LIB failed with the following output:
Change Dir: /home/pi/SoapyRTLSDR/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_d56b9/fast"
/usr/bin/make -f CMakeFiles/cmTC_d56b9.dir/build.make CMakeFiles/cmTC_d56b9.dir/build
make[1]: Entering directory '/home/pi/SoapyRTLSDR/build/CMakeFiles/CMakeTmp'
Building CXX object CMakeFiles/cmTC_d56b9.dir/src.cxx.o
/usr/bin/c++    -DHAVE_CXX_ATOMICS64_WITHOUT_LIB -std=c++11   -o CMakeFiles/cmTC_d56b9.dir/src.cxx.o -c /home/pi/SoapyRTLSDR/build/CMakeFiles/CMakeTmp/src.cxx
Linking CXX executable cmTC_d56b9
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_d56b9.dir/link.txt --verbose=1
/usr/bin/c++   -DHAVE_CXX_ATOMICS64_WITHOUT_LIB -std=c++11    -rdynamic CMakeFiles/cmTC_d56b9.dir/src.cxx.o  -o cmTC_d56b9 
/usr/bin/ld: CMakeFiles/cmTC_d56b9.dir/src.cxx.o: in function `main':
src.cxx:(.text+0x34): undefined reference to `__atomic_load_8'
/usr/bin/ld: CMakeFiles/cmTC_d56b9.dir/src.cxx.o: in function `std::__atomic_base<unsigned long long>::is_lock_free() const':
src.cxx:(.text._ZNKSt13__atomic_baseIyE12is_lock_freeEv[_ZNKSt13__atomic_baseIyE12is_lock_freeEv]+0x18): undefined reference to `__atomic_is_lock_free'
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_d56b9.dir/build.make:87: cmTC_d56b9] Error 1
make[1]: Leaving directory '/home/pi/SoapyRTLSDR/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_d56b9/fast] Error 2

Source file was:

#include <atomic>
#include <cstdint>
std::atomic<uint64_t> x (0);
int main() {
uint64_t i = x.load(std::memory_order_relaxed);
return std::atomic_is_lock_free(&x);
}

Determining if the function __atomic_load_8 exists in the atomic failed with the following output:
Change Dir: /home/pi/SoapyRTLSDR/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_e6928/fast"
/usr/bin/make -f CMakeFiles/cmTC_e6928.dir/build.make CMakeFiles/cmTC_e6928.dir/build
make[1]: Entering directory '/home/pi/SoapyRTLSDR/build/CMakeFiles/CMakeTmp'
Building CXX object CMakeFiles/cmTC_e6928.dir/CheckFunctionExists.cxx.o
/usr/bin/c++    -DCHECK_FUNCTION_EXISTS=__atomic_load_8   -o CMakeFiles/cmTC_e6928.dir/CheckFunctionExists.cxx.o -c /home/pi/SoapyRTLSDR/build/CMakeFiles/CheckLibraryExists/CheckFunctionExists.cxx
<command-line>: error: new declaration ‘char __atomic_load_8()’ ambiguates built-in declaration ‘long long unsigned int __atomic_load_8(const volatile void*, int)’ [-fpermissive]
/home/pi/SoapyRTLSDR/build/CMakeFiles/CheckLibraryExists/CheckFunctionExists.cxx:7:3: note: in expansion of macro ‘CHECK_FUNCTION_EXISTS’
   CHECK_FUNCTION_EXISTS(void);
   ^~~~~~~~~~~~~~~~~~~~~
/home/pi/SoapyRTLSDR/build/CMakeFiles/CheckLibraryExists/CheckFunctionExists.cxx: In function ‘int main(int, char**)’:
/home/pi/SoapyRTLSDR/build/CMakeFiles/CheckLibraryExists/CheckFunctionExists.cxx:17:25: error: too few arguments to function ‘long long unsigned int __atomic_load_8(const volatile void*, int)’
   CHECK_FUNCTION_EXISTS();
                         ^
make[1]: *** [CMakeFiles/cmTC_e6928.dir/build.make:66: CMakeFiles/cmTC_e6928.dir/CheckFunctionExists.cxx.o] Error 1
make[1]: Leaving directory '/home/pi/SoapyRTLSDR/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_e6928/fast] Error 2

zuckschwerdt, 2020-01-10:
`check_library_exists` can't find `atomic` in  /usr/lib/arm-linux-gnueabihf/ so I disabled the check.
It's not great but should now work as something like proof-of-concept. Please pull and try again.

wa2zkd, 2020-01-10:
and it went on to build and run

root@raspberrypi:/home/pi/SoapyRTLSDR/build# cmake ..
-- The CXX compiler identification is GNU 8.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type not specified: defaulting to release.
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Checking for module 'librtlsdr'
--   Found librtlsdr, version 0.6.0-11-gb5af
-- Found rtlsdr: /usr/local/lib/librtlsdr.so  
-- RTLSDR_INCLUDE_DIRS - /usr/local/include
-- RTLSDR_LIBRARIES - /usr/local/lib/librtlsdr.so
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB - Failed
-- Performing Test HAVE_CXX_ATOMICS64_WITH_LIB
-- Performing Test HAVE_CXX_ATOMICS64_WITH_LIB - Success
-- Performing Test HAS_STD_CXX11
-- Performing Test HAS_STD_CXX11 - Success
-- Found Git: /usr/bin/git (found version "2.20.1") 
-- Module rtlsdrSupport configured with version: 0.3.1-6186cf0
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/SoapyRTLSDR/build

zuckschwerdt, 2020-01-11:
Great! Let's close this issue and I'll clean up and finish the branch as time permits.

closed 2020-01-11 by zuckschwerdt