pothosware/SoapyRTLSDR #46: rtlsdr ppm correction, how

janvgils, 2020-03-15:
How can we correct the ppm value with soapyrtlsdr?
The current SoapySDRUtil --probe="driver=rtlsdr" isn't showing a corr=value option anymore.

`SoapySDRUtil --probe="driver=rtlsdr"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=rtlsdr
Found Rafael Micro R820T tuner
Found Rafael Micro R820T tuner

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=RTLSDR
  hardware=R820T
  origin=https://github.com/pothosware/SoapyRTLSDR
  rtl=0

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * Direct Sampling - RTL-SDR Direct Sampling Mode
       [key=direct_samp, default=0, type=string, options=(0, 1, 2)]
     * Offset Tune - RTL-SDR Offset Tuning Mode
       [key=offset_tune, default=false, type=bool]
     * I/Q Swap - RTL-SDR I/Q Swap Mode
       [key=iq_swap, default=false, type=bool]
     * Digital AGC - RTL-SDR digital AGC Mode
       [key=digital_agc, default=false, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS8, CS16, CF32
  Native format: CS8 [full-scale=128]
  Stream args:
     * Buffer Size - Number of bytes per buffer, multiples of 512 only.
       [key=bufflen, units=bytes, default=262144, type=int]
     * Ring buffers - Number of buffers in the ring.
       [key=buffers, units=buffers, default=15, type=int]
     * Async buffers - Number of async usb buffers (advanced).
       [key=asyncBuffs, units=buffers, default=0, type=int]
  Antennas: RX
  Full gain range: [0, 49.6] dB
    TUNER gain range: [0, 49.6] dB
  Full freq range: [23.999, 1764] MHz
    RF freq range: [24, 1764] MHz
    CORR freq range: [-0.001, 0.001] MHz
  Sample rates: 0.25, 1.024, 1.536, 1.792, 1.92, 2.048, 2.16, 2.56, 2.88, 3.2 MSps`

kerel-fs, 2020-03-15:
To answer a part of the question our-self: There are two possible ways to set the frequency correction via the provided SoapySDR APIs:
1.  via Frequency API by [SoapyRTLSDR::setFrequency](https://github.com/pothosware/SoapyRTLSDR/blob/8ba18f17d64005e43ff2a4e46611f8c710b05007/SoapyRTLSDR.hpp#L170-L175) with `name="CORR"`:
    ```
    const double corr_frequency = 49;  # in ppm
    std::vector<SoapySDR::Kwargs> d_tune_args;
    d_device->setFrequency(SOAPY_SDR_RX, channel, "CORR", corr_frequency, d_tune_args);
    ```
2. via Frontend corrections API by [SoapyRTLSDR::setFrequencyCorrection](https://github.com/pothosware/SoapyRTLSDR/blob/8ba18f17d64005e43ff2a4e46611f8c710b05007/SoapyRTLSDR.hpp#L142)
    ```
    const double corr_frequency = 49;  # in ppm
    d_device->setFrequencyCorrection(SOAPY_SDR_RX, channel, corr_frequency);
    ```

(Note that the underlying `rtlsdr_set_freq_correction` method supports integer values only ([include/rtl-sdr.h#L155-L162](https://github.com/osmocom/rtl-sdr/blob/dc92af01bf82b5185986190e3cde3762565d2194/include/rtl-sdr.h#L155-L162)) while both Soapy APIs accept doubles and type-casts them.)

The only question remaining:
Which of those two possibilities is preferred / do you plan to deprecate one of the two APIs in the future?

cjcliffe, 2020-06-18:
Looks like setFrequencyCorrection was added two years after the initial method of using "CORR" as a workaround was implemented.   

Use setFrequencyCorrection and consider calling setFrequency with "CORR" to be deprecated -- I'll be updating my own application to use the new function as well so that it will work with all supported devices.

closed 2021-01-25 by guruofquality