pothosware/SoapyRTLSDR #47: atomic lib linking error

miluoalbert, 2020-04-08:
When I compile SoapyRTLSDR on my Raspberry Pi 3 with GCC 8.3.0, by :
```bash
cmake -DCMAKE_INSTALL_PREFIX=/opt/install/SoapySDR  -DRTLSDR_INCLUDE_DIR=/opt/install/librtlsdr/include -DRTLSDR_LIBRARY=/opt/install/librtlsdr/lib/librtlsdr.so -DSOAPY_SDR_INCLUDE_DIR=/opt/install/SoapySDR/include -DSOAPY_SDR_LIBRARY=/opt/install/SoapySDR/lib/libSoapySDR.so ..
make -j
```
there are several errors provided by linker
```
[100%] Linking CXX shared module librtlsdrSupport.so
/usr/bin/ld: CMakeFiles/rtlsdrSupport.dir/Settings.cpp.o: in function `SoapyRTLSDR::getHardwareTime(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const':
Settings.cpp:(.text+0x12c): undefined reference to `__atomic_load_8'
/usr/bin/ld: CMakeFiles/rtlsdrSupport.dir/Settings.cpp.o: in function `SoapyRTLSDR::setHardwareTime(long long, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
Settings.cpp:(.text+0x174): undefined reference to `__atomic_store_8'
/usr/bin/ld: CMakeFiles/rtlsdrSupport.dir/Settings.cpp.o: in function `SoapyRTLSDR::setSampleRate(int, unsigned int, double)':
Settings.cpp:(.text+0x580): undefined reference to `__atomic_load_8'
/usr/bin/ld: Settings.cpp:(.text+0x5f8): undefined reference to `__atomic_store_8'
/usr/bin/ld: CMakeFiles/rtlsdrSupport.dir/Streaming.cpp.o: in function `SoapyRTLSDR::rx_callback(unsigned char*, unsigned int)':
Streaming.cpp:(.text+0x29f0): undefined reference to `__atomic_fetch_add_8'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/rtlsdrSupport.dir/build.make:131: librtlsdrSupport.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/rtlsdrSupport.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

miluoalbert, 2020-04-08:
I just fix it by add
```cmake
list(APPEND RTLSDR_LIBRARIES -latomic)
```
in to my `CMakeLists.txt`

guruofquality, 2020-04-08:
There is an open pull request to detect this: https://github.com/pothosware/SoapyRTLSDR/pull/42

Can you test if it works? I want to merge it, but I think we were waiting on some validation

miluoalbert, 2020-04-08:
> There is an open pull request to detect this: #42
> 
> Can you test if it works? I want to merge it, but I think we were waiting on some validation

Just tried it.

As for me, this pull request version worked perfectly.

------
BTW, this a lot of warnings about abi changed in GCC like:

```bash
/usr/include/c++/8/bits/vector.tcc:109:4: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<double*, std::vector<double> >’ changed in GCC 7.1
    _M_realloc_insert(end(), std::forward<_Args>(__args)...);
    ^~~~~~~~~~~~~~~~~
```
Should there be a `-Wno-psabi` flag in compile settings?

zuckschwerdt, 2020-04-08:
There is good info here: https://stackoverflow.com/questions/52020305/what-exactly-does-gccs-wpsabi-option-do-what-are-the-implications-of-supressi
TL;DR the method of passing a double type changed. Good advice not to just globally disable warnings. How many lines are affected?

zuckschwerdt, 2020-04-08:
@miluoalbert can you test current master again? The fix is now merged. I tested on ARM Buster, reopen if something doesn't work.

closed 2020-04-08 by zuckschwerdt