pothosware/SoapyRTLSDR #57: usb_claim_interface error -6

eljuligallego, 2021-05-05:
Hello I have installed the plugin for use the rtlsdr in pothos but get this error in message window and no data is received in diagram.

the rtl test is working correctly:

Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 
[R82XX] PLL not locked!
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
Allocating 15 zero-copy buffers
lost at least 80 bytes

and the SoapySDRUtil --probe="driver=rtlsdr" command seems like the wiki...

Thanks.

zuckschwerdt, 2021-05-05:
`echo 'blacklist dvb_usb_rtl28xxu' | sudo tee --append /etc/modprobe.d/blacklist-dvb_usb_rtl28xxu.conf` and restart. On Windows use make sure to properly use Zadig to uninstall the DVB-T driver.

closed 2021-05-05 by zuckschwerdt

eljuligallego, 2021-05-05:
Hi @zuckschwerdt , thanks for reply,

I have done this before due to similar problem with the rtl_test command and it solve it, but not the pothosflow issue, that it's the strange part.

I'm in ubuntu so no windows drivers problem.

guruofquality, 2021-05-05:
What SoapySDR modules are installed? I have libiio (from SoapyPlutoSDR) claim the interface and cause some trouble. Just a guess..

eljuligallego, 2021-05-05:
I have also libiio.

eljuligallego, 2021-05-06:
Hello, any option to use the SoapyRTLSDR without uninstall the pluto libiio? since I'm using it also and it give me same problem:

https://github.com/pothosware/SoapyPlutoSDR/issues/41

zuckschwerdt, 2021-05-06:
`SoapySDRUtil --info` shows the path to the `libPlutoSDRSupport.so` module, you can just move that away and if needed move it back.

eljuligallego, 2021-05-06:
Sorry but not working, tried the same with /usr/local/lib/SoapySDR/modules0.8/libPlutoSDRSupport.so but neither both solve the blocking issue.

zuckschwerdt, 2021-05-06:
So libiio is not interfering and you do not have any drivers loaded (look for dvb in `lsmod`)?
Do you have any other processes running that could lock the RTLSDR? Is there any hint in `dmesg`?

eljuligallego, 2021-05-06:
No dvb in lsmod and dmesh does hint any strange message for me:

[ 2250.278169] usb 1-4: new high-speed USB device number 5 using xhci_hcd
[ 2250.453850] usb 1-4: New USB device found, idVendor=0bda, idProduct=2838, bcdDevice= 1.00
[ 2250.453853] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2250.453854] usb 1-4: Product: RTL2838UHIDIR
[ 2250.453855] usb 1-4: Manufacturer: Realtek
[ 2250.453856] usb 1-4: SerialNumber: 00000001

I'll try with another ubuntu computer I have...

Thanks!.