pothosware/SoapyRTLSDR #59: Samplerate error

computer-wizard, 2021-11-27:
Sample rate is set to 2.28571M (2285710) in the Soapy RTLSDR Source block (GNU Radio 3.9.4).
This causes the following error,

 self.soapy_rtlsdr_source_0.set_sample_rate(0, samp_rate)
ValueError: source: Unsupported sample rate (2285710.000000).  Rate must be in the range 250000.000000, 1024000.000000, 1536000.000000, 1792000.000000, 1920000.000000, 2048000.000000, 2160000.000000, 2560000.000000, 2880000.000000, 3200000.000000
Reattached kernel driver

ncorgan, 2021-11-27:
This check was added to the GNU Radio block itself. It looks like a bug, so definitely open an issue there.

zuckschwerdt, 2021-11-27:
They likely now want to depend on `SoapyRTLSDR::getSampleRateRange()`, introduced with #54, which replaces `listSampleRates()`. But it's in the master branch only, not a tagged release for now.

computer-wizard, 2021-11-28:
OK, if is fixed in Release candidate 3.10.0.0-rc1.  Thanks for info

closed 2021-11-28 by computer-wizard