pothosware/SoapyRTLSDR #66: SoapyRTLSDR Number of samples to read

Mas313, 2023-01-20:
I have a problem reading samples from the dongle since in rx_fm there is a large number of samples (131,072) to read passed as an argument to the function as 

#define DEFAULT_BUF_LENGTH		(1 * 16384)
```
#define MAXIMUM_OVERSAMPLE		16
#define MAXIMUM_BUF_LENGTH		(MAXIMUM_OVERSAMPLE * DEFAULT_BUF_LENGTH)
samples_per_buffer = MAXIMUM_BUF_LENGTH/2
r = SoapySDRDevice_readStream(s->dev, s->stream, buffs, samples_per_buffer, &flags, &timeNs, timeoutNs);
		//fprintf(stderr, "ret=%d\n", r);

		if (r >= 0) {
			// r is number of elements read, elements=complex pairs, so buffer length in bytes is twice
			rtlsdr_callback(buf, r * 2, s);
		}
```
However, my program with same lines of code hangs until and unless i pass a very small value of samples to read from the dongle such as 512.
Can anyone tell what could be the issue? 
Also i don't see any callback function in soapyrtlsdr that i can use to get samples from rtl sdr dongle. Plz guide me if there is any.
Thanks

zuckschwerdt, 2023-01-20:
The 128k is actually the default transfer size for RTL-SDR, so not that large.
The readStream API is correct, if you want callback behavior then run the blocking call in a thread an then notify the main thread with the data frame (but know how to use threads). Otherwise, guessing that `rtlsdr_callback()` is converted librtlsdr code, this is already the wanted control flow. But maybe you'd want to loop on readStream until a full buffer is completed.
https://github.com/pothosware/SoapySDR/blob/9c4fa3241b220341479a1a76057025a8264e50d9/include/SoapySDR/Device.h#L377-L403

You omitted quite some interesting parts here. Is dev and stream setup and active? How is buffs defined? What's in timeout? Are you running that code in a loop?

Mas313, 2023-01-20:
Thanks for the prompt reply, i am running code from c example ( https://github.com/pothosware/SoapySDR/wiki/C_API_Example ) which is shown below and even with 1024 samples to read I get error which is actually related to violation when reading samples, its pointer related. I am running this code in windows 10 compiled using GCC

//create a re-usable buffer for rx samples
    complex float buff[1024];
```

    //receive some samples
    for (size_t i = 0; i < 10; i++)
    {
        void *buffs[] = {buff}; //array of buffers
        int flags; //flags set by receive operation
        long long timeNs; //timestamp for receive buffer
        int ret = SoapySDRDevice_readStream(sdr, rxStream, buffs, 1024, &flags, &timeNs, 100000);
        printf("ret=%d, flags=%d, timeNs=%lld\n", ret, flags, timeNs);
    }
```

Mas313, 2023-01-20:
Its strange that at time the code is run successfully but right now I am getting this error again when I select option to debug when receiving message that exe has stopped working.  I am running same example as mentioned above with no modification or processing of sample. you can see sample to read are just 1024.
`Unhandled exception at 0x7732F583 (ntdll.dll) in test.exe: 0xC0000005: Access violation writing location 0x00000014.`

zuckschwerdt, 2023-01-20:
It's more likely a typo in your code. The deref of a address 0x14 (=20) pointer sounds like you passed an index/length where a buffer address would be expected. Maybe push the full code to a github project, too much guesswork otherwise.

Mas313, 2023-01-20:
What i have observed now that when i handle all SOAPY errors MACROS then I no more have these issues. Thanks for the prompt replies

Mas313, 2023-01-20:
The issue seems fixed when I handle all soapysdr error macros. These were not handled in example C API and caused errors due to non handling. Thanks a lot for the prompt replies.

closed 2023-01-20 by Mas313

zuckschwerdt, 2023-01-20:
Yes, the handling of `ret <= 0` in https://github.com/pothosware/SoapySDR/wiki/C_API_Example is missing -- but that should not be an issue.
The example works for me, what did you need to add?

Mas313, 2023-01-20:
I am getting these two errors frequently reading 1024 samples. i have not tried reading a large number of samples
```
OSOAPY_SDR_OVERFLOW
SOAPY_SDR_TIMEOUT
```
Also I would be grateful if you can refer to any simple working formula for FM demodulation using these complex samples

zuckschwerdt, 2023-01-20:
I'd use FFTW and a Blackman-Harris window for a true FM demod, or otherwise an Instant Phase Angle computation for a speedy approximation.

Mas313, 2023-01-20:
can you provide  Instant Phase Angle computation for FM fm demodulation in writing or link to equations