pothosware/SoapyRTLSDR #73: Does Pothos SDR supported the RTLSDR v4 USBstick

ZephyrFanBoy, 2023-12-20:
Dear PothosWare, 

Hope your doing well.
Please confirm, I have the gut feeling Pothos SDR , soapyRTLSDR block only supported the RTLSDR v3 USBstick (R860 tuner chip. )
and not the new the RTLSDR v4 USBstick (R828D tuner chip. )
Is my cut feeling corrct?
Please advise how to make it work with the v4 version (perhaps replace *.dll somewhere)
or perhaps hint when this feature is being developed? 
Kind regards
   a SDR noob

zuckschwerdt, 2023-12-20:
The support depends on the `librtlsdr` used. In short, yes SoapySDR supports the rtlsdr blog v4 but you need an updated librtlsdr (the rtl-sdr libs with v4 support have notes to that).

closed 2023-12-20 by zuckschwerdt

ZephyrFanBoy, 2023-12-20:
Hi [zuckschwerdt](https://github.com/zuckschwerdt),

Thanks for the quick reply.
I do  not understand you, what you mean by : _you need an updated librtlsdr_ ??
I do  not understand you, what you mean by : _the rtl-sdr libs with v4 support have notes to that_??

Can you please collaborate and give a bit more help/details, because I am a noob.

zuckschwerdt, 2023-12-20:
You need a very new version of https://github.com/osmocom/rtl-sdr
It has this in August 2023: https://github.com/osmocom/rtl-sdr/commit/1261fbb285297da08f4620b18871b6d6d9ec2a7b