pothosware/SoapyRTLSDR #74: The support depends on the `librtlsdr` used. In short, yes SoapySDR supports the rtlsdr blog v4 but you need an updated librtlsdr (the rtl-sdr libs with v4 support have notes to that).

ZephyrFanBoy, 2023-12-20:
The support depends on the `librtlsdr` used. In short, yes SoapySDR supports the rtlsdr blog v4 but you need an updated librtlsdr (the rtl-sdr libs with v4 support have notes to that).

_Originally posted by @zuckschwerdt in https://github.com/pothosware/SoapyRTLSDR/issues/73#issuecomment-1864374959_

zuckschwerdt, 2023-12-20:
I guess you are asking for this to be documented. Seems a good thing.

zuckschwerdt, 2023-12-20:
Perhaps use the results of https://github.com/merbanan/rtl_433/issues/2703 for a section in the README.

ZephyrFanBoy, 2023-12-21:
Hi @zuckschwerdt 

_"I guess you are asking for this to be documented. Seems a good thing."_  Yes awesome.
Thanks very much. That's useful for me and others noobs.
because rtl-sdr.com  has a sub page called " software" 
where they explain for some SDR FOSS software how to make it work for the rtlsdrblog v4 usb stick but unfortunatly the pothos flow is not on there.
Perhaps talk to them to get it properly documented.

I am looking into the repository and results of "merbanan/rtl_433"  and "osmocom/rtl-sdr" 
I am going over the git repo's but there are no releases???
I am looking into the folders, but none have a librtlsdr.dll in there repo?

Another question where is in the pothos SDR directory the location of the librtlsdr.dll v3 that I should replace with the proper librtlsdr.dll v4?
Because searching with explorer I can not find librtlsdr.dll

zuckschwerdt, 2023-12-21:
In the `bin` directory of the Pothos install should be `rtlsdr.dll`

See https://www.rtl-sdr.com/v4/ for instructions.