pothosware/SoapyRTLSDR #79: How to build on Windows using cmake?

eabase, 2025-02-23:
Trying to build on Windows:

```bash
# cmake -S .. -B . -Ax64 -Wno-dev -Wno-deprecated --fresh --install-prefix 'C:\Program Files\SoapyRTLSDR' -DCMAKE_BUILD_TYPE=Release | Tee-object -Append $XXLOG

-- Building for: Visual Studio 17 2022
-- Selecting Windows SDK version 10.0.22621.0 to target Windows 10.0.26100.
-- The CXX compiler identification is MSVC 19.43.34808.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.43.34808/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
CMake Error at CMakeLists.txt:16 (message):
  RTL-SDR development files not found...

-- Could NOT find RTLSDR (missing: RTLSDR_LIBRARY RTLSDR_INCLUDE_DIR)
-- Configuring incomplete, errors occurred!
```

According to some sources say that VS 2022 already have `PkgConfig`...
How can I verify? 
How can I install?

What is `PKG_CONFIG_EXECUTABLE`? Is it an environment variable?

zuckschwerdt, 2025-02-23:
You could just set `RTLSDR_LIBRARY` and `RTLSDR_INCLUDE_DIR` if you don't have a working PkgConfig.

eabase, 2025-02-24:
I've tried to populate those with all sorts of paths and files, and it's not working. I need to see someone's working definitions for those. 

In addition there seem to be some confusion between `RTLSDR_INCLUDE_DIRS` and `RTLSDR_INCLUDE_DIR` (no "S"). I also tried to get a working pkgconfig, but I keep on getting error, that the project is not configured correctly for using **vcpkg**...

Finally, you say here: 
https://github.com/pothosware/SoapyRTLSDR/wiki

> The CMake build system will locate librtl-sdr and SoapySDR development files on your system. The plugin module will be built and installed into the SoapySDR modules directories.

```bash
git clone https://github.com/pothosware/SoapyRTLSDR.git
cd SoapyRTLSDR
mkdir build
cd build
cmake ..

make   <------------------- There are NO Makefiles in the project!
sudo make install <-------- ...So how do you build this on Windows?
```

It would be great if you guys actually tested this E2E before writing up instructions that can't possibly work as posted. Especially for very low engagement project like this.

zuckschwerdt, 2025-02-24:
Yes, there is regular confusion about the PkConfig names. Best to lookup what's used in the [CMakeLists.txt](https://github.com/pothosware/SoapyRTLSDR/blob/master/CMakeLists.txt#L18)
Here it is actually `RTLSDR_INCLUDE_DIRS` and `RTLSDR_LIBRARIES`, sorry about that.

On Linux or MacOS this would work: `cmake -DRTLSDR_INCLUDE_DIRS=/usr/local/include -DRTLSDR_LIBRARIES=/usr/local/lib -B build && cmake --build build`

I don't think there are active Windows developers in the SoapySDR project, your insight and documentation will be very useful.

zuckschwerdt, 2025-02-24:
With CMake you can select the "generator", for Windows it should be `-G"Visual Studio 17 2022"`, e.g. (adjust the paths) `cmake -G"Visual Studio 17 2022" -DRTLSDR_INCLUDE_DIRS=C:\rtlsdr\include -DRTLSDR_LIBRARIES=C:\rtlsdr\lib -B build && cmake --build build`

eabase, 2025-02-24:
Yes, I know about the build command, but I don't know what are in those locations. It seem that evyone have different ideas what is a *library*.

For example, looking in the (downloadable) package for `libusb-1.0.27`, we have the following files:

```bash
$ tree -L 4
.
├── README.txt
├── VS2022
│   ├── MS32
│   │   ├── dll
│   │   │   ├── libusb-1.0.dll
│   │   │   ├── libusb-1.0.exp
│   │   │   ├── libusb-1.0.lib
│   │   │   └── libusb-1.0.pdb
│   │   └── static
│   │       ├── libusb-1.0.lib
│   │       └── libusb-1.0.pdb
│   └── MS64
│       ├── dll
│       │   ├── libusb-1.0.dll
│       │   ├── libusb-1.0.exp
│       │   ├── libusb-1.0.lib
│       │   └── libusb-1.0.pdb
│       └── static
│           ├── libusb-1.0.lib
│           └── libusb-1.0.pdb
├── include
│   └── libusb.h
└── libusb-1.0.def
```

So what would the `RTLSDR_INCLUDE_DIRS` and `RTLSDR_LIBRARIES`, point at?

OC, if you compile the `libusb` or `rtl-sdr` yourself, you get a completely different structure and locations...

eabase, 2025-02-24:
BTW. There is a [libusb-cmake](https://github.com/libusb/libusb-cmake) version as well... In case someone else is also having issues. (I didn't try it yet.)

eabase, 2025-02-24:
Great. What about the filenames themselves?

For example, I noticed that `rtl-sdr` produces `rtlsdr.dll` whereas some *include* criteria from other packages, expect to to load `librtlsdr.dll`, or something like it, etc.

eabase, 2025-02-24:
Aaah may bad!  Sorry!

Never mind, I'm confsing the usblib with rtl-sdr above. 
Hold on.

eabase, 2025-02-24:
> So `-DRTLSDR_INCLUDE_DIRS=include -DRTLSDR_LIBRARIES=VS2022\MS64\dll`

Ok so this is wrong. 

The compiler option variables are without 'S', like this:  
**`-DRTLSDR_INCLUDE_DIR`**  and
**`-DRTLSDR_LIBRARY`**

So using this, worked:
`-DRTLSDR_INCLUDE_DIR="$RTLSDR_INCLUDE_DIRS" -DRTLSDR_LIBRARY="$RTLSDR_LIBRARIES"`

I am using the [rtl-sdr-blog](https://github.com/rtlsdrblog/rtl-sdr-blog), but had to copy the *.h files from repo, like this: 

```bash
# tree -C -L 3 C:\mylib\rtl-sdr

C:\mylib\rtl-sdr
├── from_rtl-sdr-blog.txt
├── include
│   ├── reg_field.h
│   ├── rtl-sdr.h
│   ├── rtl-sdr_export.h
│   ├── rtlsdr_i2c.h
│   ├── tuner_e4k.h
│   ├── tuner_fc0012.h
│   ├── tuner_fc0013.h
│   ├── tuner_fc2580.h
│   └── tuner_r82xx.h
└── lib
    ├── convenience_static.lib
    ├── libgetopt_static.lib
    ├── msvcr100.dll
    ├── pthreadVC2.dll
    ├── rtl_adsb.exe
    ├── rtl_biast.exe
    ├── rtl_eeprom.exe
    ├── rtl_fm.exe
    ├── rtl_power.exe
    ├── rtl_sdr.exe
    ├── rtl_tcp.exe
    ├── rtl_test.exe
    ├── rtlsdr.dll
    ├── rtlsdr.exp
    ├── rtlsdr.lib
    └── rtlsdr_static.lib

```

Here's the build output:

```bash
# $XXLOG='log_install_soapysdr.txt'
# $RPATH='C:\mylib\rtl-sdr'
# $RTLSDR_INCLUDE_DIRS="$RPATH\include"
# $RTLSDR_LIBRARIES="$RPATH\lib"

# cmake -S .. -B . -Ax64 -Wno-dev -Wno-deprecated --fresh --install-prefix 'C:\Program Files\SoapyRTLSDR' -DCMAKE_BUILD_TYPE=Release -DRTLSDR_INCLUDE_DIR="$RTLSDR_INCLUDE_DIRS" -DRTLSDR_LIBRARY="$RTLSDR_LIBRARIES" | Tee-object -Append $XXLOG

-- Building for: Visual Studio 17 2022
-- Selecting Windows SDK version 10.0.22621.0 to target Windows 10.0.26100.
-- The CXX compiler identification is MSVC 19.43.34808.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.43.34808/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- [INFO] SoapySDR Development files found.
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
-- Found RTLSDR: C:/mylib/rtl-sdr/lib
-- CMAKE_MODULE_PATH   : C:/Program Files/SoapySDR/cmake;C:/mydev/gitclones/SoapyRTLSDR
[... 69 lines cut ...]
Call Stack (most recent call first):
  CMakeLists.txt:73 (SOAPY_SDR_MODULE_UTIL)

-- Generating done (0.1s)
-- Build files have been written to: C:/mydev/gitclones/SoapyRTLSDR/build

#---------------------------------------------------------------------------------------------------------
# cmake --build . --config Release --parallel 8
#---------------------------------------------------------------------------------------------------------

MSBuild version 17.13.9+e0f243f1e for .NET Framework

  1>Checking Build System
  Building Custom Rule C:/mydev/gitclones/SoapyRTLSDR/CMakeLists.txt
  Registration.cpp
  Settings.cpp
  Streaming.cpp
  Version.cpp

Registration.obj : error LNK2019: unresolved external symbol __imp_rtlsdr_get_device_count referenced in function "class std::vector<class std::map<class std::basic_string<char,struct std::char_traits<char>
...

# LOTS OF ERRORS

```

eabase, 2025-02-24:
Ok, so `RTLSDR_LIBRARIES` actually have to point to the (`*.dll`) file, so now it makes sense that the compiler option is called `RTLSDR_LIBRARY`.

Using: `$RTLSDR_LIBRARIES="C:\mylib\rtl-sdr\lib\rtlsdr.dll"`
It now builds ok, but fails linking:

```bash
# cmake -S .. -B . -Ax64 -Wno-dev -Wno-deprecated --fresh --install-prefix 'C:\Program Files\SoapyRTLSDR' -DCMAKE_BUILD_TYPE=Release -DRTLSDR_INCLUDE_DIR="$RTLSDR_INCLUDE_DIRS" -DRTLSDR_LIBRARY="$RTLSDR_LIBRARIES"

-- Building for: Visual Studio 17 2022
-- Selecting Windows SDK version 10.0.22621.0 to target Windows 10.0.26100.
-- The CXX compiler identification is MSVC 19.43.34808.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.43.34808/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- [INFO] SoapySDR Development files found.
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
-- Found RTLSDR: C:/mylib/rtl-sdr/lib/rtlsdr.dll
-- CMAKE_MODULE_PATH   : C:/Program Files/SoapySDR/cmake;C:/mydev/gitclones/SoapyRTLSDR
-- RTLSDR_INCLUDE_DIRS : C:/mylib/rtl-sdr/include
-- RTLSDR_LIBRARIES    : C:/mylib/rtl-sdr/lib/rtlsdr.dll
-- Looking for rtlsdr_set_bias_tee
-- Looking for rtlsdr_set_bias_tee - not found
-- Looking for rtlsdr_set_dithering
-- Looking for rtlsdr_set_dithering - not found
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.48.1.windows.1")
-- Module rtlsdrSupport configured with version: 0.3.3-f22516c
-- Configuring done (3.9s)
-- Generating done (0.3s)
-- Build files have been written to: C:/mydev/gitclones/SoapyRTLSDR/build

# cmake --build . --config Release --parallel 8

MSBuild version 17.13.9+e0f243f1e for .NET Framework

  1>Checking Build System
  Building Custom Rule C:/mydev/gitclones/SoapyRTLSDR/CMakeLists.txt
  Registration.cpp
  Settings.cpp
  Streaming.cpp
  Version.cpp

C:\mylib\rtl-sdr\lib\rtlsdr.dll : fatal error LNK1107: invalid or corrupt file: 
cannot read at 0x358 [C:\mydev\gitclones\SoapyRTLSDR\build\rtlsdrSupport.vcxproj]

```

eabase, 2025-02-24:
ok, so now we have `SoapyRTLSDR`, which is then needed by `SoapySDR`, which is then needed by `LimeSuitNG`... 🤯

We need to cleanup the all `CMakeLists.txt` files a bit to provide more useful and clear info.

eabase, 2025-02-24:
Yes, something is off in the install location...

I see I kept specifying it as build/compile option, as "other" projects often kept installing  to the wrong folder (`C:\Program Files (x86)\` instead of `C:\Program Files\`). But that was probably wrong, here.

```bash
# tree -C -L 4 "C:\Program Files\SoapyRTLSDR"

C:\Program Files\SoapyRTLSDR
└── lib
    └── SoapySDR
        └── modules0.8-3
            └── rtlsdrSupport.dll
```

Problem is that installer insists prefixing the path with `/lib` as in: `/lib/SoapySDR/modules0.8-3/`.

Where is that picked up?

eabase, 2025-02-24:
Changing to `--install-prefix 'C:\Program Files\SoapySDR\'` gives this:

```bash
# cmake --build . --config Release --target install

MSBuild version 17.13.9+e0f243f1e for .NET Framework

  rtlsdrSupport.vcxproj -> C:\mydev\gitclones\SoapyRTLSDR\build\Release\rtlsdrSupport.dll
  1>
  -- Install configuration: "Release"
  -- Installing: C:/Program Files/SoapySDR/lib/SoapySDR/modules0.8-3/rtlsdrSupport.dll

# tree -C -L 3 "C:\Program Files\SoapySDR\lib\"

C:\Program Files\SoapySDR\lib\\
├── SoapySDR
│   └── modules0.8-3
│       └── rtlsdrSupport.dll 
```

The running `SoapySDRUtil`:

```bash
SoapySDRUtil.exe --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-g309335ec
API Version: v0.8.200
ABI Version: v0.8-3

Install root: C:\Program Files\SoapySDR
Search path:  C:\Program Files\SoapySDR/lib/SoapySDR/modules0.8-3
Module found: C:\Program Files\SoapySDR/lib/SoapySDR/modules0.8-3/rtlsdrSupport.dll
  LoadLibrary() failed: The specified module could not be found.

Available factories... No factories found!
Available converters...
...
```

This is weird. 
1. Notice how the `Module found` while the `LoadLibrary` fails. 
2. Notice how there are mixed `\` and `/` in the paths.

eabase, 2025-02-24:
@zuckschwerdt 
Any idea what this means?

eabase, 2025-02-24:
Seem like `rtlsdr.dll` is missing... So it have been be dynamically compiled?

```bash
ldd rtlsdrSupport.dll | grep -iE --color=always "not found|$"
```

[image]

Copying the `rtlsdr.dll` from my "library":
```powershell
cd 'C:\Program Files\SoapySDR\lib\SoapySDR\modules0.8-3\'
cp C:\mylib\rtl-sdr\lib\rtlsdr.dll .

# SoapySDRUtil.exe --probe

######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device
Found Rafael Micro R820T tuner
[INFO] Opening Generic RTL2832U OEM :: 00000001...
Found Rafael Micro R820T tuner

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=RTLSDR
  hardware=R820T
  index=0
  origin=https://github.com/pothosware/SoapyRTLSDR

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: YES
  Time sources: sw_ticks
  Other Settings:
...
```

Great! BUT... 

There is a bug in the `SoapySDRUtil.exe`. 
It shows the wrong case of the currently loaded driver, and therefore misinform you about how to type the probe command. 

```bash
# SoapySDRUtil.exe --probe="driver=rtlsdr"     # <------------------------  OK !

######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=rtlsdr
Found Rafael Micro R820T tuner
[INFO] Opening Generic RTL2832U OEM :: 00000001...
Found Rafael Micro R820T tuner

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=RTLSDR
  hardware=R820T
  index=0
  origin=https://github.com/pothosware/SoapyRTLSDR

# ======================================================================================
# Whereas if you actually use capital (uppercase) letter for the same, you get an error!
# SoapySDRUtil.exe --probe="driver=RTLSDR"    # <------------------------  NOT ok, ERROR !

######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=RTLSDR
Error probing device: SoapySDR::Device::make() no match
```

eabase, 2025-02-24:
Also, the `Sample rates` I get, are completely different (and seemingly wrong) than what's shown in the Wiki page: https://github.com/pothosware/SoapyRTLSDR/wiki#detecting-the-rtl-sdr

`MINE`: Sample rates: **[0.225001, 0.3], [0.900001, 3.2] MSps**
`Wiki`: Sample rates: **0.25, 1.024, 1.536, 1.792, 1.92, 2.048, 2.16, 2.56, 2.88, 3.2 MSps**

zuckschwerdt, 2025-02-24:
The "factories" (`SoapySDR::Registry`, what you use for the `driver=` selection) for the module is different from the driver key meta data (`getDriverKey()`). SoapySDRUtil shows the meta data not the modules registered factory. Not user-friendly, agreed.

No idea how to statically compile on MSVC, sorry. But it's not recommended anyway, easier to swap or update individual libs.

Sample rates are read from the rtlsdr lib's `rtlsdr_get_sample_rate()` and might depend on the hardware and particular lib.

eabase, 2025-02-25:
> No idea how to statically compile on MSVC, sorry. But it's not recommended anyway, 

The only reason why it is *not recommended* is that people are too lazy to fix anything for Windows, and also refuse to document anything if they actually do anything. So I'm not sure what you mean there, unless you have some expert additional knowledge on why it is not recommended?

> easier to swap or update individual libs.

Swap to what!? I assume they also need to get compiled into static.

Anyway, I've been trying to compile this for 3 days now, and it seem I'm the only one in the universe wanting this, so clearly all other SDR users & developer out there, is using something else. I almost lost track of why I even wanted to do this. The idea was to create a pre-packaged python project for an RTL-SDR and LimeSDR-USB device, and using GNUradio. I'm nowhere near getting there as I can't even get the most basic stuff working. Which is very weird, because the entire framework for all these projects are available for download for native Windows. So surely someone knows how to do this.